septillion-git/FadeLed #2: PWM with 1024 steps (10 bit)

sebastian3107, 2016-11-16:
Which parts do I have to change to get a 10 bit resolution?
I want to use this library to control an RGB strip with an ESP8266 board.

septillion-git, 2016-11-16:
Good question / idea. Is it alright if I'll have a look this weekend?

From the top of my head you have to change the variable types of all the variables that hold the brightness from byte to unsigned int or uint16_t. And also rewrite the overflow and underflow detection in the update to match. I think that's all but I'll have a look this weekend. Let's see if I can make it easy to vary :D

septillion-git, 2016-11-21:
As of version v1.3.0 it's now possible to use 10-bit PWM :) Simply set it in FadeLed.h

closed 2016-11-21 by septillion-git

sebastian3107, 2016-11-21:
Great work! Thank you very much! :+1:

septillion-git, 2016-11-21:
I must say I didn't test it on a 10-bit but it should work. At least didn't break the 8-bit so should be fine :p Just open a new issue if you experience a bug.

bedenko, 2017-08-28:
i tried to compile the fadeLedBasic example on NodeMCU.
Changed  **FADE_LED_PWM_BITS** to **10**, but i get following error:

`/home/user/Arduino/libraries/FadeLed/src/FadeLed.cpp: In constructor 'FadeLed::FadeLed(byte)':

/home/user/Arduino/libraries/FadeLed/src/FadeLed.cpp:13:45: error: cannot convert 'const uint16_t* {aka const short unsigned int*}' to 'const flvar_t* {aka const unsigned int*}' in initialization
                             _biggestStep(100){
                                             ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).`

septillion-git, 2017-09-05:
Sorry for the late response!

Just started a new job so my time is a bit limited at the moment. It's indeed due to the different size of int between 8-bit ATmega's and the ESP. I will change that in a future release 👍 But for now I don't really have the time to sit down and make e proper release. 😢