septillion-git/FadeLed #10: FadeLed licence ?

machliang, 2017-03-07:
Hi,
     I can not find any licence about this project, did I lost some pages?
     If could, this project licence can be "MIT" or "Apache" ? 
     Thanks a lot~
                          Mars Liang

machliang, 2017-03-08:
or 3-clause BSD ? 
                            Mars Liang

septillion-git, 2017-03-08:
Yep, I should add a license. I mainly skipped that for now because of the work involved reading about it. 👼  I will add a license soon. I'm thinking MIT or LGPL for now

bmoren, 2021-01-22:
Any updates on adding a License to this project?

here's an easy spot to help choose: https://choosealicense.com/

Thanks!

septillion-git, 2021-01-29:
Hi Ben, sorry for the slow response. Something about long days working from home... Hope you're doing well too.

Still find the whole licensing a bit messy 👼 But thanks for the link! Although I really like how short the MIT license is, I do like the copyleft concept. So I was thinking GPLv3. Would that work in your project as well?

bmoren, 2021-01-29:
no problem! 

yes GPL3 is great! I use that for many of my projects. It's good to add a license since technically no-one is allowed to use your project without one as you retain all copyright and usage. As far as I know even including a library that's not attached to an open source license in a project is a misuse of the software (using copyright material w/o permission).

septillion-git, 2021-01-31:
Okay, I will add it tomorrow 😄

septillion-git, 2021-02-01:
GPLv3 in place!

closed 2021-02-01 by septillion-git