septillion-git/FadeLed #9: const uint16_t FadeLedGammaTable[101]  in FadeLedGamma.h

machliang, 2017-03-07:
Hi, 
    This is a nice project, thanks~
    While I am trying to compile the example for my ESP8266(10 bit PWM resolution) , 
    here was a compilation error :

Arduino/libraries/FadeLed/src/FadeLed.cpp:13:45: error: cannot convert 'const uint16_t* {aka const short unsigned int*}' to 'const flvar_t* {aka const unsigned int*}' in initialization
                             _biggestStep(100){

    After some experiments and source code tracing, I suggest to update the lines in FadeLedGamma.h :
          const uint16_t FadeLedGammaTable[101] PROGMEM = {....
                         to 
          const flvar_t FadeLedGammaTable[101] PROGMEM = {....
                if FADE_LED_PWM_BITS >= 9
    to avoid the compilation error. 
    Thanks a lot!
                         Mars Liang

septillion-git, 2017-03-08:
I will look into it 😃 I'm really a AVR user and didn't test the latest version on a ESP.

Thanks for telling me!

machliang, 2017-03-08:
You are welcome~

commit 5bc727a5 (2017-11-28): Fixed #9 Changed FadeLedGammaTable type to flvar_t
commit 1858b56f (2017-11-28): Fixed #9 Changed FadeLedGammaTable type to flvar_t
closed 2017-11-28 by septillion-git