septillion-git/FadeLed #14: gamma.py is mentioned in docs, but not part of repo

myhd, 2017-10-17:
Let's say someone wants to have 512 steps in the gamma correction list. The docs say that "you can use the provided Python script. […] Call it like: `python gamma.py` etc…"

I do suspect the lack of file is an oversight, right?

septillion-git, 2017-10-18:
Ahh, yeah. I was ahead of myself 😃 I'll add it in a future. (Still needs some work and my repo is a bit of a mess 😳 )

For now, here's the Python script: [gamma.zip](https://github.com/septillion-git/FadeLed/files/1394129/gamma.zip)

Btw, for 512 steps to make sense you need at least 12-bit (preferably more) PWM. 101 steps on 8-bit is actually already pushing the limits. But a percentage just is easy to understand. 👼

myhd, 2017-10-18:
Thank you!

I am using 16bit PWM on an Arduino Duemilanove (It is possible and very satisfying. See [StackExchange article](https://arduino.stackexchange.com/q/12718/21566)) with a eight-second fade. The original 100 steps you have provided are quite noticeable with this setup. The script will help.

One aside question: Why are you using a value table, and not compute the values as they are needed? To save cycles? I found this [article about Exponential LED Fading](https://diarmuid.ie/blog/pwm-exponential-led-fading-on-arduino-or-other-platforms/) quite interesting.

septillion-git, 2017-10-18:
If you have 16-bit then it's possible. Although I think 255 steps would already satisfy our eyes. But maybe with a higher gamma.

101 steps is a trade off on 8-bit. Steps on 8-bit are to big and will always be noticeable.

And I use a look up because the calculation is pretty intensive. Especially because I want this to run on 8-bit as well. And even on 32-bit it's still some work. One objective of the Python script is that only step 0 may result in a duty cycle of 0. Step 1 must result in the first possible brightness.

I'll read the article tonight when I'm home. Now back to work 👼

septillion-git, 2017-10-18:
Read it and yeah, it's close to what I do here. But I used a real Gamma factor but you can probably work that out from his as well. But here (in the python script) you can really set it.

Also, I added the precondition that step 1 may not result in 0% duty cycle (aka a dark led).

Doing a log10() on an Arduino, is not simple. And on a 8-bit unit very very very slow. So some data in a lookup is way faster :) I _might_ implement an extrapolated lookup table version to save memory but I don't really think it's worth it because it's already saved in program memory.

septillion-git, 2018-04-27:
Added the scripts in 15d4ad414e4e298ac847d3fc50349de8bbe10e1a

closed 2018-04-27 by septillion-git