septillion-git/FadeLed #21: ESP32 support

franzuu, 2018-10-28:
Hi, I must say, what a great library! I have used it with ESP8266 and it worked great. 
But can you please make it compatible with ESP32?
I am currently getting this error, while using Atom + PlatformIO:
lib\FadeLed\src/FadeLed.h:20:28: fatal error: avr/pgmspace.h: No such file or directory

Thank you in advance!

septillion-git, 2018-10-28:
Thanks!

But yeah, support for other micro's isn't that great. Mostly because I don't use them as much. Would like to but yeah, time...

But I thought I could just add ESP32 next to the ESP8266 support. But man! What a can of worms! 😩  Indeed it has no analogWrite() function... So supporting it isn't that simple. I think I can do that somewhere this week. Think I'll just use ledcWrite() instead of analofWrite() on an ESP32. So it will not take care of the rest of the ledc stuff. Should not be to hard but it needs some changes in the code, library.properties, add a new example to show how it works and find an ESP32 to test it... 😓 

I'll keep you posted!

septillion-git, 2018-12-02:
Mm, as you might have noticed this will take me a little longer... Still would really like to include ESP32 support!

franzuu, 2018-12-03:
Hi! Sorry for late reply! I am super thankful that you are planning to add ESP32 support! Take your time, no need to rush things. As I was writing this issue last time, I was trying out new things and had a lot other stuff to do also. That was the reason I forgot to answer. But I still feel this library is the way to go! Works really well with esp8266! If you need I can test the code with esp32 after you have made changes and report back to you.

joelv1964, 2018-12-04:
Hi Septillion-git, I come to join the request of franzuu. It would be really cool if FadeLed could be adapted to the ESP32. It is really a useful library that releases the main loop for other treatments and brings a real comfort of use for the user. I am currently working on a model car wifi control and FadeLed simulates the inertia of filaments of incandescent bulbs. I think ESP32 will be more and more present in object applications. You do a great job. If you manage to adapt FadeLed to the ESP32, it will be really SUPER. Congratulations in advance!

septillion-git, 2019-01-19:
@franzuu @joelv1964 
I made an attempt to add simple ESP32 support. See the branch eb527daf22f176ce6a6a5a5aeb2917119ad84295. 

It's a little bit different than on an Uno or ESP8266 because you pass a ledc channel rather than a pin. FadeLed does not assign a ledc channel. 1) Because that was simpler to implement. But 2) because doing it in the library would not give you insight in which ledc channels are used otherwise. This would make the library less transparent. I added an example (FadeLedESP32) to "show" how to use it.

And I say "show" because besides compiling I was unable to test it 🙄 I can't find my ESP32... I will order one but if one would like to test it, great! Do you know how to install the version from the branch?

Couple of questions: 
- Does it work?
- What is the default pin of the LED on which board?
- And if you really want to get exotic, does 16-bit mode work? (for this you do have to edit stuff. Might make an example later.

joelv1964, 2019-01-19:
Hi, I'm happy to read your answer. I have installed the version from the branch. I just tried on an ESP32, it's great! The example works perfectly. On my ESP32 board LEDbuilt in on GPIO1, but my board is Chinese cheap 🙄...
I don't need the 16-bit mode but maybe others will need it. I quickly see if all the functions work like on the ESP8266.
You did a great job!

Joel

septillion-git, 2019-01-21:
That's great news!

I'll try to make a 16-bit example this week as well. If that's fine (can you test that for me? Still waiting for my ESP32) I'll try to make a release this week. :smile: 

And nothing wrong with China ESP32. That's what I buy. Can you show me which version you have? Because would make sense to pick a pin which has already an LED connected.

joelv1964, 2019-01-21:
Hi,

No problem for 16-bit PWM test 😀. 
I saw that according to the manufacturer of the ESP32 board, the LED_BUILTIN could be connected to GPIO1, GPIO2 on many boards, or GPIO9.

My ESP32 boards are [this model](https://m.aliexpress.com/item/32828869161.html?pid=808_0000_0109&spm=a2g0n.search-amp.list.32828869161&aff_trace_key=c7e2f7b572504f019f3d658f705805a6-1548084650007-00850-UneMJZVf&aff_platform=msite&m_page_id=8708amp-dEsXkHyqoCnflSlDpja7dw1548084666288&gclid=EAIaIQobChMIrsW825j_3wIVUETTCh2FOwvKEAAYASAAEgJfAvD_BwE). 

Have a Good day

septillion-git, 2019-01-22:
After struggling a while I made an 16-bit example. Why was I struggling? Because I made an error last time! I never set the PWM to 10-bit (in FadeLed.h) like I said so it was initialized to 8-bit. And you showed that also works fine 😁 See a4f0521dd36af91d84192a9f34ab08664ab77e6b

But okay, fixed that an added an example in the branch for 16-bit. I'll probably make all gamma tables (all bit depths) available in code when I release it. Makes it even easier to change the bit depth (no need to generate new table for the 2,3 gamma).

New test: c18f9e2f1233405512654e0b9f22b6e3df205c23 Also use LED_BUILTIN in it. Which board do you select in the IDE to make it work?

joelv1964, 2019-01-22:
Hi,
Good news: with errors FadeLed works! :joy:
I will not be able to test your new code before this week, but I will test it!
In my IDE, the board selected is "ESP32 Dev Module".
Do you want to publish a single FadeLed version compatible ESP8266 / ESP32 or 2 different versions?
Joël

septillion-git, 2019-01-22:
That's fine! I might change the example again 👼 

It's going to be a single release 😄 That's most of the work, keep it compatible with everything 😋 I also made some changes in the master branch. So this is going to be 1.6.0 if all done and tested 👍 .

I'll keep you updated!

joelv1964, 2019-01-22:
Great.
I could not wait this week to test.
I am experiencing a compilation error :worried:: "'LED_BUILTIN' was not declared in this scope".
I think that we should define the pin2 as output directly in the example, "LED_BUILTIN" is obviously not recognized by the ESP32.

septillion-git, 2019-01-22:
Yeah, I saw that error now as well. The ESP32 Dev Module has no LED_BUILTIN. I committed changes to the branch again including a fix for the error. See 9015743d9e1673ac0744a328856f15ff92d1a3b1

Whoooohooo, 100 commits 😋

joelv1964, 2019-01-22:
OK, I tested the new example. I just replaced "LED_BUILTIN" with "1" for my board.
It's perfect :ok_hand:
The fade is more clean with 10-Bit Pwm compared to 8-Bit Pwm.
The result is the same as on ESP8266.
Many Thanks :wink:

joelv1964, 2019-01-28:
Hi,

How are you?
I had a problem last night 😟.
I need 8 FadeLed channels in the same project with an ESP32.

The ESP32 has a curious behavior. Only 5 channels made correctly with FadeLed.
I think I understood, but I did not have time to try.
In FadeLed.h, at line 63, there is "#define FADE_LED_MAX_LED 6".
On the ESP32, 16 channels are usable (0 to 15). Should not be Fade_LED_MAX_LED to "16" in the case of using an ESP32?

#if defined (ESP8266)
     #define FADE_LED_MAX_LED 6
#else
     #define FADE_LED_MAX_LED 16
#end if

Tanks in advance for your reply.

septillion-git, 2019-01-28:
Ahh, yeah, the max led problem. Not only an issue for the ESP32 even, you can also have it on a ESP8266 now I think about it. But it has less pins so it's less likely. But the mean reason to limit it is memory usage. An ESP8266 or ESP32 is less critical than a Uno or ATtiny but think I'll try it to do it neat:

```C++
#ifndef FADE_LED_MAX_LED
  #if defined(ESP8266)
    #define FADE_LED_MAX_LED  14
  #elif defined(ESP32)
    #define FADE_LED_MAX_LED  16
  #elif defined(ARDUINO_AVR_MEGA2560) || defined(ARDUINO_AVR_MEGA)
    #define FADE_LED_MAX_LED  15
  #else
    #define FADE_LED_MAX_LED  6
  #endif
#endif
```

I think max ever on a ESP8266 is 14 pins. And this also increases it for a Mega (both 2560 and 1280 which I think both have max 15 PWM pins). Probably there still are types which support more but default to less but also the other way around 🙄 It's a start 😄

septillion-git, 2019-01-28:
Made a commit including the changes (including case for Uno/Nano/Pro Mini.
d5197aa6abc9c8aefe32a1af272b46eba4403d92

joelv1964, 2019-01-28:
Cool :smile:,
I try this new code in 2 or 3 days.
I will tell you if it works properly.
Bye

septillion-git, 2019-01-28:
No problem! Good to get the bugs out before release anyway 😄

septillion-git, 2019-02-06:
Great! All I need is some time to go over all the documentation :roll_eyes: Hope I can do that this weekend.

septillion-git, 2021-02-01:
Wow, I really thought I already did all this! Should implement this!

littlej956, 2022-06-19:
Hi, any update on this ?