septillion-git/FadeLed #24: Constrain PWM range

SparkyDan555, 2019-06-16:
Is it possible to constrain the PWM range  by setting max / min values? For example if user sets min PWM of 255 and max PWM to 1023 (10 bit PWM) and they request 0, it would output 0 but if they request 1 the output would be 255 then every value between 2 and 1023 outputs accordingly - possibly with map()? This would be useful for fading objects that have a 'dead zone' where they need to warm up like halogen bulbs. Unlike LED's, halogens dont start glowing until approx 20% duty cycle.

septillion-git, 2019-06-17:
Hi,

Yeah, you could do that by making your own Gamma table. This is basically a lookup table for the input. So if you make a table like:
```C++
const flvar_t myGammaTable[769] PROGMEM = {0, 255, 256, 257, 258, ...};
```
Only now you lost gamma correction. You can fix this by making a gamma table with to many steps and chop off the part up to 255. For example
```C++
/* 211-step brightness table: gamma = 2.3 for 10-bit PWM*/ 
const flvar_t myGammaTable[101] PROGMEM = {
	  /*0,   1,   1,   1,   1,   1,   2,   2,   2,   3,
	  3,   3,   4,   4,   5,   5,   6,   6,   7,   8,
	  8,   9,  10,  11,  12,  13,  13,  14,  15,  16,
	 17,  19,  20,  21,  22,  23,  25,  26,  27,  29,
	 30,  32,  33,  35,  37,  38,  40,  42,  44,  45,
	 47,  49,  51,  53,  55,  57,  59,  62,  64,  66,
	 69,  71,  73,  76,  78,  81,  83,  86,  89,  92,
	 94,  97, 100, 103, 106, 109, 112, 115, 118, 122,
	125, 128, 132, 135, 139, 142, 146, 149, 153, 157,
	161, 164, 168, 172, 176, 180, 184, 188, 193, 197,
	201, 206, 210, 214, 219, 223, 228, 233, 237, 242, */
	247, 252, 257, 262, 267, 272, 277, 282, 288, 293,
	298, 304, 309, 315, 320, 326, 332, 338, 343, 349,
	355, 361, 367, 373, 380, 386, 392, 398, 405, 411,
	418, 424, 431, 438, 444, 451, 458, 465, 472, 479,
	486, 493, 500, 508, 515, 522, 530, 537, 545, 552,
	560, 568, 576, 584, 591, 599, 607, 616, 624, 632,
	640, 649, 657, 665, 674, 683, 691, 700, 709, 718,
	727, 735, 745, 754, 763, 772, 781, 791, 800, 809,
	819, 829, 838, 848, 858, 868, 878, 887, 898, 908,
	918, 928, 938, 949, 959, 970, 980, 991, 1001, 1012,
	1023
};
```
This was originally a table with 211 steps, 2.3 gamma for 10-bit. But I'm not sure if this will look good...

You could also try to generate a table for 10-bit with some extra steps and increment them all with 254 and chop off the top above 1023.
```C++
const flvar_t myGammaTable[101] PROGMEM = {
  0, 255, 255, 256, 256, 257, 258, 258, 259, 261
  262, 263, 265, 266, 268, 270, 272, 274, 276, 278
  281, 283, 286, 289, 292, 295, 299, 302, 306, 310
  314, 318, 322, 326, 331, 336, 341, 346, 351, 356
  362, 368, 374, 380, 386, 392, 399, 406, 412, 420
  427, 434, 442, 450, 458, 466, 474, 483, 491, 500
  509, 519, 528, 538, 548, 558, 568, 578, 589, 600
  611, 622, 633, 645, 657, 669, 681, 693, 706, 718
  731, 745, 758, 772, 785, 799, 814, 828, 843, 857
  872, 888, 903, 919, 935, 951, 967, 983, 1000, 1017,
  1023
};
```

Hope that's useful :smile:

SparkyDan555, 2019-06-20:
Thanks! I made a 101 step gamma table with index 0 at 0, index 1 at 175, index 2 at 183 etc... all the way up to index 100 at 1023. It works great. It would still be nice to see this implemented into the main code so a custom lookup table doesnt have to be worked out and the user can constrain the output easily and can be easily adjusted. 🤔

septillion-git, 2019-06-26:
Sorry for the late reply!

But what you ask is pretty specific. So I don't think I'll add it to the library. Especially because that would create a certain need to maintain it for all the number of resolutions etc.

What I can (/might) do if I come across more of these is to add them into the documentation. As a sort off "as is" example list of custom "gamma" tables.

closed 2022-07-06 by SparkyDan555