septillion-git/FadeLed #25: fade leds via Shift Register? (e.g. 74HC595)

perfectom, 2019-12-16:
Hi, please is there a possibility to use fadeled against a shift register to fade leds ? Or would this be a new feature? 

Thanks! -Tom

septillion-git, 2019-12-17:
Hi,

Only option to dim leds connected to a shift register is to use software to soft-PWM them. FadeLed is based around hardware PWM of common micro controllers. So unfortunately it's not possible to use it with shift registers.

I still want to branch off a "FadeAll" version of the library that you can use to fade every function. But that will need some work to do it right which I don't have at the moment. (Aka, to many projects and no need for FadeAll myself at this moment 👼 ).

If you really want to you can make a copy of FadeLed and use something like [ShiftRegister-PWM](https://github.com/Simsso/ShiftRegister-PWM-Library) (which funny enough is from a different Timo 😄 ) to do the dimming part. But do note, the dimming/PWM of the shift register outputs does take up quite some resource of the micro controller.

perfectom, 2019-12-18:
Hi Timo! Thanks for the information there!  I might try your suggestion to use ShiftRegister-PWM over the holidays. FadeLed is great, by the way! Thanks a lot for creating it :)

closed 2021-02-01 by septillion-git