septillion-git/FadeLed #26: Inverting the output

dougp2, 2020-12-22:
Thanks for the Library!

Would it be difficult to add an inversion feature/capability to the library?  What I mean is, fading on/off seems to assume  an Arduino output being driven high to light a common cathode LED.  If the LED is common anode fade on/off must be reversed to show the desired effect.

septillion-git, 2020-12-22:
Thanks!

Don't think it's worth adding it as a dedicated feature but it's already possible now. Just mirror the desired gamma table and use that table. This way the output is mirrored as well 😃

dougp2, 2020-12-22:
Yes, reversing the table order inverts the fade effect.  Thanks!  This still leaves sines.on() and sines.off() acting backwards.  Is a workaround possible with #define?  Or, something else I could do without modifying the library?

septillion-git, 2020-12-23:
.on() and .off() are just shorthand for .set(0) and .set(max) so they should also flip when you use a mirrored gamma table.

dougp2, 2020-12-25:
.on() and .off() are just shorthand for .set(0) and .set(max) so they should also flip when you use a mirrored gamma table.
==========+=======

Yes, it is as you say.  Thanks again!